jean-luc1203/jkbms-rs485-addon #3: Two BMS devices shows as one

Rosentorp, 2025-06-15:
Hi!

I have a system with two JK BMS's. The first one is connected to a Victron Cerbox GX (and a Multiplus II) via CAN. Between the two BMSes there is a RS-485 link. The first one has address zero (0) with all DIP-switches off. The second one har address 1 with the first DIP-switch on.

Everything works fine with the Cerbo GX and all shows up fine. ALso if I connect with the JK BMS app all systems shows fine (if you select address 0 in the app to show all BMSes or directly to BMS 1 with address 1).

Using the addon it shows two BMS as set up but both shows the same info:

[image]

[image]

- path: /dev/ttyUSB1
  nb_jkbms: 2
  use_gateway: false
  gateway_ip: ""
  gateway_port: 5000

mqttadresse: ****
mqttport: 1883
mqttuser: mqtt-user
mqttpass: ****

Rosentorp, 2025-06-16:
Address 1 and 2 works as you says. To be able to work with Victron Cerbo it need to be one in master (0) and on at address one (1)....

jean-luc1203, 2025-06-17:
The batteries must start their address at 1, then 2, etc. The software uses address 0 and is the master of the Modbus bus.
I don't have a Victron, but I gather that it uses an address larger than 16.

jean-luc1203, 2025-06-28:
I made a mistake in my previous comment. In fact, when the JKBMS is set to position 0, it becomes the master of the bus and changes mode, talking to the other BMSs and regularly sending bus information. In this mode, you can no longer change any parameters, unless you place a command when the master JK is doing nothing. There can only be one master on a bus.

closed 2025-07-12 by jean-luc1203