jean-luc1203/jkbms-rs485-addon #4: Monitoring fails when a master battery is present in the system

agapitoangel, 2025-06-16:
I've observed an issue where the system works correctly only when no battery is configured as master. However, if a battery is set to master mode, monitoring fails or is limited.

This is problematic for real-world installations because:

 - Victron inverters require a master battery to 
   operate correctly in multi-battery setups.

  - When one battery is in master mode, the system appears to monitor only that battery, ignoring the rest of the bank.

- This leads to incomplete and potentially dangerous monitoring, as the inverter may receive data from just one battery while the actual state of the full battery bank remains unknown.

As it stands, the system appears to be compatible only with single-battery setups, and is not safe for battery banks with a master unit.

jean-luc1203, 2025-06-17:
On a Modbus bus, there can only be one master. 
It is the master that queries the slaves. The software is the master and by default it takes address 0. 
The various BMSs must therefore be set to other values. 
I have a site where there are 4 JKBMSs on the bus and everything is OK (The switches are set to 1, 2, 3, 4). 
All the values from each BMS are reported correctly. 
Interval of less than 5 seconds. This allows Home Assistant to respond with automated actions if necessary.

jean-luc1203, 2025-07-12:
Solved in v2.0.1

closed 2025-07-12 by jean-luc1203