jean-luc1203/jkbms-rs485-addon #5: Soon I will add an RS485-IP gateway

jean-luc1203, 2025-06-17:
I am currently adding the ability to connect an RS485-IP gateway of this type.
I had it lying around somewhere. This means you can have the PC anywhere and no longer need to connect a physical USB port. 
It’s starting to work. In a few days, I will make this version available on my GitHub.

This is the one i’m using

[image]

ensjoridder, 2025-06-29:
> Thank you.
> Could you tell me more about your proposal? I confess I still don't understand the idea.

Great work on getting the RS485-IP gateway up and running! That’s a solid and professional solution, especially for industrial setups. If you ever want to experiment with alternatives, my idea was to use a Raspberry Pi Zero with `ser2net` and ZeroTier—essentially turning the Pi into a flexible, low-cost serial-to-IP bridge with built-in VPN security. It’s not as robust as your gateway but offers some fun customization options. Really impressive how you’re solving this challenge!

jean-luc1203, 2025-06-29:
Ah yes, I understand your idea now. 
I particularly like the idea of encrypting data from one point to another. 
With Node Red, it's actually quite easy to implement. If there's demand for it, we could start creating a flow. It will definitely work on a Pi Zero. 
Ah, but I used Zerotier for a long time until I switched to Tailscale, which is more granular and based on an open standard.
When shall we do it?  😉👍

ensjoridder, 2025-06-29:
Thank you for the kind words and enthusiasm—it means a lot! 😊 To be fully open, I don’t have deep theoretical knowledge of this myself. I’ve been learning by leveraging tools like ChatGPT, along with insights from GitHub projects and YouTube tutorials. It’s been a mix of experimenting and adapting ideas to fit my needs. For example, I successfully used the concept I proposed (with `ser2net`, ZeroTier, and `com0com`) to remotely monitor and control a TinyBMS S516 from Enepaq. Now, I’m working on building more battery packs with JK-BMS modules, which will be connected in parallel. It’s been a fun but challenging journey, and I’m still learning as I go. I must admit, **Tailscale** is somewhat new to me, but it sounds like a great tool for secure remote communication. Similarly, I haven’t worked much with **Node-RED** yet, but it seems incredibly powerful and flexible—especially for integrating with MQTT and Home Assistant. Both sound exciting to explore! If you’re interested, we could collaborate or share ideas—I’d love to learn from your experience while contributing what I can if it could be helpful 🙂

ensjoridder, 2025-06-29:
> Ah yes, I understand your idea now. 💡
> I particularly like the idea of encrypting data from one point to another.
> With Node Red, it's actually quite easy to implement. If there's demand for it, we [...]

Thank you for the kind words and enthusiasm—it means a lot! 😊 To be fully open, I don’t have deep theoretical knowledge of this myself. I’ve been learning by leveraging tools like ChatGPT, along with insights from GitHub projects and YouTube tutorials. It’s been a mix of experimenting and adapting ideas to fit my needs. For example, I successfully used the concept I proposed (with `ser2net`, ZeroTier, and `com0com`) to remotely monitor and control a TinyBMS S516 from Enepaq. Now, I’m working on building more battery packs with JK-BMS modules, which will be connected in parallel. It’s been a fun but challenging journey, and I’m still learning as I go. I must admit, **Tailscale** is somewhat new to me, but it sounds like a great tool for secure remote communication. Similarly, I haven’t worked much with **Node-RED** yet, but it seems incredibly powerful and flexible—especially for integrating with MQTT and Home Assistant. Both sound exciting to explore! If you’re interested, we could collaborate or share ideas—I’d love to learn from your experience while contributing what I can if it could be helpful 🙂

jeybee74, 2025-06-29:
I have a few Waveshare rs485 ETH devices laying around and will test this soon😊

[image]