jean-luc1203/jkbms-rs485-addon #7: Jkbms adresse

jeremyd62, 2025-06-22:
Hello, and congratulations on this work, which greatly facilitates my task of recovering my batteries under HA.

However, I am encountering a minor issue.
Is there a way to read a jkbms with an address of 0 (I am required to set this battery to 0 so that my UPS recognises both batteries, otherwise it only sees one)?

jeremyd62, 2025-06-25:
The deye communicates with the master battery by the can, it only reads the news and does not send an order

If the master battery is not on address 0, it cannot read the slave batteries and considers that it only has the capacity of the master battery

jean-luc1203, 2025-06-28:
I've done some research into how JKBMS work
It's absolutely essential that there is no address set to 0 on one of the JKBMSs, as in this case it becomes the bus master and my software, which is also the master, can no longer retrieve data from the BMSs. 
This is how JK works for inverters linked in CAN mode. They don't ask questions, they just periodically receive data from the JK master.
The BMS that has become the master regularly broadcasts data from the JKBMS present, and no longer allows questions to be put to it.
This poses a problem if your inverter requires it to operate in this way. Have you tried connecting it to the other rs485 port on the left of the board?
I have connected my WKS Ultra to this port and the information is retrieved by it.