jean-luc1203/jkbms-rs485-addon #10: !!  Warning  !!

jean-luc1203, 2025-06-28:
### !!  Warning  !!
I released a new version yesterday, 
it was to fix the fact that JK setups were getting lost when upgrading. It worked locally during my tests.
I realized there was a **problem when I updated** the Github with this version. 
So I removed the v1.2.5 version and went back to v1.2.0
I suggest you go back to v1.2.0 if possible.
Time for me this Sunday to take a closer look at what's going on. 
**Sorry for the inconvenience**

jean-luc1203, 2025-06-28:
I have just released version 2.1.6 which should work. I wasn't able to test it with the pc's USB port, just with a gateway. I may test it tomorrow.

jeybee74, 2025-06-28:
Works with USB-RS485 also👍

But I have tried a few converters, some work and some does not work, seems like the ones that shows up in the hardware list as ttyACMx does not work, only the ones that shows as ttyUSBx.

Both types works with other integrations, like the one for my Sunsynk inverter.

jean-luc1203, 2025-06-29:
Thank you @jeybee74  for the information and test result. 
Could you give me a few brands that it doesn't work with? I could try to get one to test.
I'll leave you to open another issue on this subject.

closed 2025-06-29 by jean-luc1203