jean-luc1203/jkbms-rs485-addon #17: Do we have to be in `bms_broadcasting: false` mode to be able to change settings of the BMS ?

jean-luc1203, 2025-07-21:
> Do we have to be in `bms_broadcasting: false` mode to be able to change settings of the BMS ? 

 _Originally posted by @psychotik2k3 in [#16](https://github.com/jean-luc1203/jkbms-rs485-addon/issues/16#issuecomment-3097599939)_

jean-luc1203, 2025-07-21:
Yes, exactly. It is essential that no BMS is in the MASTER position, so all 4 switches must be in position 0000.

Each BMS must have an address from 1 to a maximum of 15. So in your case, set the switches to "1000"..

⚠️The SETUP section is still in Beta test. It does not include all the protections found in the JKBMS application. ⚠️

psychotik2k3, 2025-07-22:
My bms got ID 1 (0001 on dipswitch) I'll check that I can really chance
values from the device screen on home assistant. Because I tried but values
were not changed.
Is there anything to do before? Or to validate the change of the value?

Thanks

brycelarge, 2025-07-30:
Hi @jean-luc1203 what happens to CAN output to the inverter when all BMS are in slave mode and your code is in master mode. I would think master BMS job is to send SOC, charge settings and SOC back to the inverter via CAN and now we replacing master how is the CAN communication effected to the inverter?

psychotik2k3, 2025-07-30:
> Hi [@jean-luc1203](https://github.com/jean-luc1203) what happens to CAN output to the inverter when all BMS are in slave mode and your code is in master mode. I would think master BMS job is to send [...]

For me he added the possibility to connect through can bus as well as rs485 for those who for some reason prefer to connect on can.
This way you choose how to connect depending on your configuration.

But this integration is meant to dialog directly with the jk bms not through any inverter.
There are a lot of configuration possible  and his works allows most of them.
I really thanks him for his fantastic job.

brycelarge, 2025-07-31:
> > > [@brycelarge](https://github.com/brycelarge) and [@psychotik2k3](https://github.com/psychotik2k3)
> > > In my opinion, CAN communication has no interference whatsoever with what happens on the RS485 BUS and whether the BMS is master or slave.
> > > On the CAN, the data from the BMS (and only it, not the others) is sent repeatedly every x thousandths of a second. I don't know if there's a way to send commands to the BMS via the CAN. All I know is how to read via this bus. Which is absolutely not the case on the rs485.
> > > In my software, if you enable CAN communication, it disables RS485 communication. I could leave it active but that doesn't seem logical. In any case, I haven't had any requests along these lines. Thank you both for your appreciation of my work. I really appreciate it.
> > 
> > 
> > I think I might not have explained it well enough. If you remove master out of the RS485 communication and have your code acting as master over RS485 then the job master had which was to know the state of all connected BMS which it gathers over RS485 communication, then calculate the total SOC, SOH, Capacity and relay information back to the inverter by means of CAN communication. So if you have 2 X 100ah batteries one is at 50% SOC and the other at 60% SOC then it calculates 55% SOC and sends that via CAN to the inverter.
> > I will test this out once I get an ESP RS485 wifi gateway working.
> > Thanks anyway for the hard work, this is exactly what I was looking
> 
> For me when using multiple bms connected in serial, one master and severally others, the connection to the can bus only tell the inverter of a global battery, with a mutualised SOC.
> 
> Afterward only the bmses handle the different charge and discharge among all bms. The rôle of master bms is basically to compute the global soc, and setting 100% only when all bms charged at 100%
> 
> But in the inverter side there is no change, it communicantes with a bms

Yes exactly the job of the master BMS is to compute the global state to the inverter. So I was just wondering what happens when the software is in master mode and there is no global compute anymore.