jean-luc1203/jkbms-rs485-addon #18: No MQTT devices are created

ElCorristo, 2025-08-05:
Hi,

I've already invested many hours, but I can't create MQTT devices from the Mosquitto in HA. Perhaps one of you can help me. That would be great, as I have no idea where my problem could be.

I have four batteries with JK-BMS. One battery is configured as the master and connected to the Deye inverter via CAN bus.

The master is connected to the HA's NUC (HA-OS) via an RS485-USB adapter.

According to the instructions, I configured the JK-BMS add-on as follows:

jkbms:

cfi700, 2025-08-06:
I have the same problem.
I have two batteries, one is set as master (ID:0000) the second batterie  is set to ID:0001
Communication to the inverter is via CAN.
Communication to HA is via RS485

Please see the log:

> start
> node $NODE_OPTIONS node_modules/node-red/red.js --settings /etc/node-red/config.js

✅ config.js is being loaded
✅ Node-RED is using userDir = /config/smartphoton_jkbms/
✅ Node-RED is using uiPort = 1891
6 Aug 12:02:34 - [info] 

Welcome to Node-RED
===================

6 Aug 12:02:34 - [info] Node-RED version: v4.1.0
6 Aug 12:02:34 - [info] Node.js  version: v22.16.0
6 Aug 12:02:34 - [info] Linux 6.12.35-haos x64 LE
6 Aug 12:02:34 - [info] Loading palette nodes
6 Aug 12:02:34 - [info] Node-RED Contrib Theme Collection version: v4.1.0
6 Aug 12:02:35 - [info] Dashboard version 3.6.5 started at /endpoint/ui
6 Aug 12:02:36 - [info] node-red-contrib-home-assistant-websocket v0.77.2 nodes initialized
6 Aug 12:02:36 - [info] Settings file  : /etc/node-red/config.js
6 Aug 12:02:36 - [info] Context store  : 'default' [module=localfilesystem]
6 Aug 12:02:36 - [info] User directory : /config/smartphoton_jkbms/
6 Aug 12:02:36 - [warn] Projects disabled : editorTheme.projects.enabled=false
6 Aug 12:02:36 - [info] Flows file     : /config/smartphoton_jkbms/flows.json
6 Aug 12:02:36 - [warn] Encrypted credentials not found
6 Aug 12:02:36 - [info] Server now running at http://127.0.0.1:1891/
6 Aug 12:02:36 - [info] Starting flows
6 Aug 12:02:36 - [info] [ui-base:Setup] Node-RED Dashboard 2.0 (v1.26.0) started at /dashboard
6 Aug 12:02:36 - [info] [ui-base:Setup] Created socket.io server bound to Node-RED port at path /endpoint/dashboard/socket.io
6 Aug 12:02:36 - [info] Started flows
6 Aug 12:02:36 - [error] [serialconfig:6e1abcbcc6ebb714] serial port /dev/JKBMS error: Error: Error: No such file or directory, cannot open /dev/JKBMS
6 Aug 12:02:36 - [info] [mqtt-broker:56f7b2737cce493b] Connection failed to broker: mqtt://core-mosquitto.local.hass.io:1883
6 Aug 12:02:37 - [info] [mqtt-broker:56f7b2737cce493b] Connected to broker: mqtt://192.168.10.130:1883
6 Aug 12:02:37 - [info] [function:Définir global variables] Variable globale 'path' définie: /dev/serial/by-id/usb-1a86_USB_Single_Serial_558E118098-if00
6 Aug 12:02:37 - [info] [function:Définir global variables] Variable globale 'nb_jkbms' définie: 2
6 Aug 12:02:37 - [info] [function:Définir global variables] Variable globale 'use_gateway' définie: false
6 Aug 12:02:37 - [info] [function:Définir global variables] Variable globale 'gateway_ip' définie: 192.168.100.11
6 Aug 12:02:37 - [info] [function:Définir global variables] Variable globale 'gateway_port' définie: 5000
6 Aug 12:02:37 - [info] [function:Définir global variables] Variable globale 'bms_broadcasting' définie: true
6 Aug 12:02:37 - [info] [function:Définir global variables] Variable globale 'CAN_bus_usage' définie: false
6 Aug 12:02:37 - [info] [function:Définir global variables] Variable globale 'mqttadresse' définie: 192.168.10.130
6 Aug 12:02:37 - [info] [function:Définir global variables] Variable globale 'mqttport' définie: 1883
6 Aug 12:02:37 - [info] [function:Définir global variables] Variable globale 'mqttuser' définie: xxxxxx
6 Aug 12:02:37 - [info] [function:Définir global variables] Variable globale 'mqttpass' définie: xxxxxxxxx
6 Aug 12:02:37 - [info] [function:Définir global variables] Variable globale 'ssl' définie: false
6 Aug 12:02:37 - [info] [function:Définir global variables] Variable globale 'certfile' définie: fullchain.pem
6 Aug 12:02:37 - [info] [function:Définir global variables] Variable globale 'keyfile' définie: privkey.pem
6 Aug 12:02:37 - [info] serial port /dev/JKBMS closed
6 Aug 12:02:37 - [info] [serialconfig:6e1abcbcc6ebb714] serial port /dev/serial/by-id/usb-1a86_USB_Single_Serial_558E118098-if00 opened at 115800 baud 8N1

Configuration:
jkbms:
  jkbms_path: /dev/serial/by-id/usb-1a86_USB_Single_Serial_558E118098-if00
  jkbms_count: 2
  use_gateway: false
  gateway_ip: 192.168.100.11
  gateway_port: 5000
  bms_broadcasting: true
  CAN_bus_usage: false
mqtt:
  mqttadresse: 192.168.10.130
  mqttport: 1883
  mqttuser: xxxx
  mqttpass: xxxxxxxx
ssl: false
certfile: fullchain.pem
keyfile: privkey.pem

Also in MQTT-Explorer no devices are shown.

jean-luc1203, 2025-08-06:
so the MQTT link is OK. It's the RS485 dialogue that's obviously not working.
Can you show us which adapter you're using? Does it have 2 LEDs, one for reception and one for transmission?
If so, are the 2 flashing?
How have you wired it?
Have you inverted A and B for testing ?
Is GND connected?

ElCorristo, 2025-08-06:
I use this RS485-USB adapter from Waveshare.

[image]

I've previously connected it to my Raspberry Pi from the Solar Assistant, and it worked flawlessly. The A and B cables are connected correctly. GND is not connected.

On the LEDs, only the RX one is flashing.

jean-luc1203, 2025-08-06:
What bothers me is that Linux HAOS sees it as ACM0 and not USBx. Doesn't it treat it with the same driver as USBx?
You're right about broadcasting, I misread and misinterpreted it. Translation isn't always easy. 
Is it possible to change the adapter?

ElCorristo, 2025-08-06:
HA shows me the following information about the USB connection with the adapter under Hardware:

ttyACM0
/dev/serial/by-id/usb-1a86_USB_Single_Serial_5958016438-if00
Subsystem:
tty
Gerätepfad:
/dev/ttyACM0
ID:
/dev/serial/by-id/usb-1a86_USB_Single_Serial_5958016438-if00
Attribute:
DEVLINKS: >-
  /dev/serial/by-id/usb-1a86_USB_Single_Serial_5958016438-if00
  /dev/serial/by-path/pci-0000:00:14.0-usb-0:3:1.0
  /dev/serial/by-path/pci-0000:00:14.0-usbv2-0:3:1.0
DEVNAME: /dev/ttyACM0
DEVPATH: /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/tty/ttyACM0
ID_BUS: usb
ID_MODEL: USB_Single_Serial
ID_MODEL_ENC: USB\x20Single\x20Serial
ID_MODEL_ID: 55d3
ID_PATH: pci-0000:00:14.0-usb-0:3:1.0
ID_PATH_TAG: pci-0000_00_14_0-usb-0_3_1_0
ID_PATH_WITH_USB_REVISION: pci-0000:00:14.0-usbv2-0:3:1.0
ID_REVISION: '0445'
ID_SERIAL: 1a86_USB_Single_Serial_5958016438
ID_SERIAL_SHORT: '5958016438'
ID_TYPE: generic
ID_USB_DRIVER: cdc_acm
ID_USB_INTERFACES: ':020201:0a0000:'
ID_USB_INTERFACE_NUM: '00'
ID_USB_MODEL: USB_Single_Serial
ID_USB_MODEL_ENC: USB\x20Single\x20Serial
ID_USB_MODEL_ID: 55d3
ID_USB_REVISION: '0445'
ID_USB_SERIAL: 1a86_USB_Single_Serial_5958016438
ID_USB_SERIAL_SHORT: '5958016438'
ID_USB_TYPE: generic
ID_USB_VENDOR: 1a86
ID_USB_VENDOR_ENC: 1a86
ID_USB_VENDOR_ID: 1a86
ID_VENDOR: 1a86
ID_VENDOR_ENC: 1a86
ID_VENDOR_ID: 1a86
MAJOR: '166'
MINOR: '0'
SUBSYSTEM: tty
TAGS: ':systemd:'
USEC_INITIALIZED: '684841100358'

I don't currently have an alternative adapter available. I use three of these adapters and haven't had any problems so far.

I could order a different one online, but that would take a while. Should I use a specific adapter? Do you have a recommendation?

jean-luc1203, 2025-08-06:
Come to think of it, since you're using the CAN port for the inverter, could you set each BMS to a different address and, above all, not set one as master? And change the broadcasting parameter to false.
I'd like to see if there's anything in the log this time.
Well, it's past midnight here. I'm going to call it a night. I'll check back tomorrow to see what happens next.

cfi700, 2025-08-07:
> so the MQTT link is OK. It's the RS485 dialogue that's obviously not working. Can you show us which adapter you're using? Does it have 2 LEDs, one for reception and one for transmission? If so, are the 2 flashing? How have you wired it? Have you inverted A and B for testing ? Is GND connected? Did-you change the paramater "bms_broadcasting" to false ?

Only A and B is connected. No GND.
The RxD LED is flashing regularly.
The TxD LED remains off.

I also use the Adapter from Waveshare successfully with the JK-BMS PC-Ssoftware to set up the two BMS. 
A second identical one is connected to Solar-Assistant (Raspberry PI) and the Inverter for communication with no problems.

[image]

HA hardware listing:
[image]

I set both of the BMs to induvidual adresses (no master) and "bms_broadcasting" to false as you suggested.
Now i get this messages in the log:

7 Aug 07:47:31 - [info] [serialconfig:6e1abcbcc6ebb714] serial port /dev/ttyACM0 opened at 115800 baud 8N1
7 Aug 07:47:32 - [error] [function:Vérifie Trame 2] TypeError: Cannot read properties of undefined (reading 'length')
7 Aug 07:47:32 - [error] [function:Vérifie Trame 3] TypeError: Cannot read properties of undefined (reading 'length')
7 Aug 07:47:34 - [error] [function:Vérifie Trame 3] TypeError: Cannot read properties of undefined (reading 'length')
7 Aug 07:47:38 - [error] [function:Vérifie Trame 3] TypeError: Cannot read properties of undefined (reading 'length')
7 Aug 07:47:42 - [error] [function:Vérifie Trame 3] TypeError: Cannot read properties of undefined (reading 'length')

ElCorristo, 2025-08-07:
I just ordered two different adapters with FTDI FT232R chips. They'll be delivered tomorrow. I'll test them then.

Thanks to jean-luc1203 for your great support and have a nice vacation.

jean-luc1203, 2025-08-07:
great ! Wainting for tomorrow news.

I can't go into "broadcasting" mode because I obviously don't have access to the switches to switch to "0000". 
So I'll have to wait until next week before I can log the rs485 port response.

Thanks, it's a nice & quiet holiday 🚠

cfi700, 2025-08-07:
Hello, I have great news!
After replacing the "old" (ttyACMx) USB-RS485 dongle with one that shows ttyUSBx in System/Hardware, communication now works properly.

[image]

Thanks you jean-luc1203 for your great support. Your suggestion to change the USB adapter at least solved my problem.

I'm curious if this was also ElCorristo's problem.

ElCorristo, 2025-08-08:
With the new adapter it works right away.

[image]

I won't get around to setting it up until this evening. But I'll definitely buy the dashboard as a small thank you.

ElCorristo, 2025-08-09:
@cfi700 

According to your screen, you're getting a different number of entities generated or delivered with your two BMSs. 

[image]

Could you please check which entities are delivered more frequently with BMS 1 than with BMS Master?

cfi700, 2025-08-09:
@ElCorristo 
Yes, that's true.
I hadn't noticed it before. Because all the entities I use for the GUI appear in both "devices."

Nevertheless, I looked for differences and found the following:
In BMS_1, there are entities, e.g.,
BMS BMS
or BMS Brand
that are not included in BMS_Master.

I think this is a relic from my attempts to establish a connection via RS485. These additional entities were also only sent once.

The update interval seems to be the same for both for the "normal" entities.

ElCorristo, 2025-08-09:
@cfi700 

Thanks for the quick reply.

Which GUI are you using? Or did you create something yourself?

cfi700, 2025-08-10:
@ElCorristo 

It's a combination of the dashboard from
jean-luc1203 and a few modifications by me.
For example, the combined SOC across all batteries.

[image]