jean-luc1203/jkbms-rs485-addon #20: Error in installation

mcexc, 2025-08-10:
Could not install on HAOS. Error log as below:

Docker build failed for f7107e65/aarch64-addon-smartphoton_jkbms:3.0.7 (exit code 1). Build output: #0 building with "default" instance using docker driver #1 [internal] load build definition from Dockerfile #1 transferring dockerfile: 3.23kB done #1 WARN: NoEmptyContinuation: Empty continuation line (line 67) #1 DONE 0.0s #2 [internal] load metadata for ghcr.io/hassio-addons/base:stable #2 DONE 0.5s #3 [internal] load .dockerignore #3 transferring context: 2B done #3 DONE 0.0s #4 [1/7] FROM ghcr.io/hassio-addons/base:stable@sha256:90f3b27a296bf5603a1113943199815a19d45f829adfe1c28e7a24e0b4faaee5 #4 DONE 0.0s #5 [internal] load build context #5 transferring context: 531.05kB 0.0s done #5 DONE 0.0s #6 [2/7] COPY package.json /opt/ #6 CACHED #7 [3/7] COPY node-red-dashboard-show-dashboard.patch /tmp/ #7 CACHED #8 [4/7] WORKDIR /opt #8 CACHED #9 [5/7] RUN apk add --no-cache --virtual .build-dependencies build-base linux-headers py3-pip python3-dev && apk add --no-cache git icu-data-full nginx nodejs npm openssh-client patch can-utils iproute2 bash mosquitto-clients && npm config set fetch-timeout 300000 && npm config set fetch-retry-mintimeout 20000 && npm config set fetch-retry-maxtimeout 120000 && npm config set fetch-retries 5 && npm config set registry https://registry.npmjs.org/ && npm install --no-audit --no-fund --no-update-notifier --omit=dev --unsafe-perm --fetch-timeout=300000 --fetch-retry-mintimeout=20000 --fetch-retry-maxtimeout=120000 && npm rebuild --build-from-source @serialport/bindings-cpp && npm cache clear --force && echo -e "StrictHostKeyChecking no" >> /etc/ssh/ssh_config && patch -d /opt/node_modules/node-red-dashboard -p1 -i /tmp/node-red-dashboard-show-dashboard.patch && apk del --no-cache --purge .build-dependencies && rm -fr /etc/nginx /root/.cache /root/.npm /root/.nrpmrc /tmp/* #9 0.171 fetch https://dl-cdn.alpinelinux.org/alpine/v3.22/main/aarch64/APKINDEX.tar.gz #9 0.368 fetch https://dl-cdn.alpinelinux.org/alpine/v3.22/community/aarch64/APKINDEX.tar.gz #9 0.880 (1/44) Installing libgcc (14.2.0-r6) #9 0.895 (2/44) Installing jansson (2.14.1-r0) #9 0.904 (3/44) Installing libstdc++ (14.2.0-r6) #9 0.987 (4/44) Installing binutils (2.44-r2) #9 1.291 (5/44) Installing libmagic (5.46-r2) #9 1.348 (6/44) Installing file (5.46-r2) #9 1.353 (7/44) Installing libgomp (14.2.0-r6) #9 1.372 (8/44) Installing libatomic (14.2.0-r6) #9 1.377 (9/44) Installing gmp (6.3.0-r3) #9 1.401 (10/44) Installing isl26 (0.26-r1) #9 1.480 (11/44) Installing mpfr4 (4.2.1_p1-r0) #9 1.510 (12/44) Installing mpc1 (1.3.1-r1) #9 1.520 (13/44) Installing gcc (14.2.0-r6) #9 5.803 (14/44) Installing libstdc++-dev (14.2.0-r6) #9 6.170 (15/44) Installing musl-dev (1.2.5-r10) #9 6.401 (16/44) Installing g++ (14.2.0-r6) #9 7.749 (17/44) Installing make (4.4.1-r3) #9 7.766 (18/44) Installing fortify-headers (1.1-r5) #9 7.772 (19/44) Installing patch (2.8-r0) #9 7.784 (20/44) Installing build-base (0.5-r3) #9 7.784 (21/44) Installing linux-headers (6.14.2-r0) #9 7.957 (22/44) Installing libbz2 (1.0.8-r6) #9 7.965 (23/44) Installing libexpat (2.7.1-r0) #9 7.975 (24/44) Installing libffi (3.4.8-r0) #9 7.983 (25/44) Installing gdbm (1.24-r0) #9 7.991 (26/44) Installing xz-libs (5.8.1-r0) #9 8.007 (27/44) Installing mpdecimal (4.0.1-r0) #9 8.020 (28/44) Installing libpanelw (6.5_p20250503-r0) #9 8.025 (29/44) Installing sqlite-libs (3.49.2-r1) #9 8.099 (30/44) Installing python3 (3.12.11-r0) #9 8.815 (31/44) Installing python3-pycache-pyc0 (3.12.11-r0) #9 9.193 (32/44) Installing pyc (3.12.11-r0) #9 9.193 (33/44) Installing py3-setuptools-pyc (80.9.0-r0) #9 9.362 (34/44) Installing py3-pip-pyc (25.1.1-r0) #9 9.567 (35/44) Installing py3-parsing (3.2.3-r0) #9 9.586 (36/44) Installing py3-parsing-pyc (3.2.3-r0) #9 9.610 (37/44) Installing py3-packaging-pyc (25.0-r0) #9 9.626 (38/44) Installing python3-pyc (3.12.11-r0) #9 9.626 (39/44) Installing py3-packaging (25.0-r0) #9 9.639 (40/44) Installing py3-setuptools (80.9.0-r0) #9 9.748 (41/44) Installing py3-pip (25.1.1-r0) #9 9.914 (42/44) Installing pkgconf (2.4.3-r0) #9 9.925 (43/44) Installing python3-dev (3.12.11-r0) #9 11.31 (44/44) Installing .build-dependencies (20250810.053407) #9 11.32 Executing busybox-1.37.0-r18.trigger #9 11.35 OK: 362 MiB in 76 packages #9 11.46 fetch https://dl-cdn.alpinelinux.org/alpine/v3.22/main/aarch64/APKINDEX.tar.gz #9 11.83 fetch https://dl-cdn.alpinelinux.org/alpine/v3.22/community/aarch64/APKINDEX.tar.gz #9 12.66 (1/28) Installing can-utils (2025.01-r0) #9 12.70 (2/28) Installing pcre2 (10.43-r1) #9 12.73 (3/28) Installing git (2.49.1-r0) #9 13.04 (4/28) Installing git-init-template (2.49.1-r0) #9 13.05 (5/28) Installing icu-data-full (76.1-r1) #9 14.11 (6/28) Installing libcap2 (2.76-r0) #9 14.11 (7/28) Installing libelf (0.193-r0) #9 14.12 (8/28) Installing libmnl (1.0.5-r2) #9 14.13 (9/28) Installing iproute2-minimal (6.15.0-r0) #9 14.16 (10/28) Installing libxtables (1.8.11-r1) #9 14.17 (11/28) Installing iproute2-tc (6.15.0-r0) #9 14.20 (12/28) Installing iproute2-ss (6.15.0-r0) #9 14.20 (13/28) Installing iproute2 (6.15.0-r0) #9 14.23 Executing iproute2-6.15.0-r0.post-install #9 14.23 (14/28) Installing cjson (1.7.18-r1) #9 14.24 (15/28) Installing mosquitto-libs (2.0.21-r0) #9 14.25 (16/28) Installing mosquitto-clients (2.0.21-r0) #9 14.26 (17/28) Installing nginx (1.28.0-r3) #9 14.27 Executing nginx-1.28.0-r3.pre-install #9 14.31 Executing nginx-1.28.0-r3.post-install #9 14.32 (18/28) Installing ca-certificates (20250619-r0) #9 14.36 (19/28) Installing ada-libs (2.9.2-r4) #9 14.38 (20/28) Installing icu-libs (76.1-r1) #9 14.55 (21/28) Installing simdjson (3.12.0-r0) #9 14.56 (22/28) Installing simdutf (7.2.1-r0) #9 14.57 (23/28) Installing nodejs (22.16.0-r2) #9 16.19 (24/28) Installing npm (11.3.0-r0) #9 16.46 (25/28) Installing openssh-keygen (10.0_p1-r7) #9 16.48 (26/28) Installing libedit (20250104.3.1-r1) #9 16.49 (27/28) Installing openssh-client-common (10.0_p1-r7) #9 16.57 (28/28) Installing openssh-client-default (10.0_p1-r7) #9 16.61 Executing busybox-1.37.0-r18.trigger #9 16.62 Executing ca-certificates-20250619-r0.trigger #9 16.68 OK: 474 MiB in 104 packages #9 17.01 npm warn Unknown builtin config "globalignorefile". This will stop working in the next major version of npm. #9 17.01 npm warn Unknown builtin config "python". This will stop working in the next major version of npm. #9 17.29 npm warn Unknown builtin config "globalignorefile". This will stop working in the next major version of npm. #9 17.29 npm warn Unknown builtin config "python". This will stop working in the next major version of npm. #9 17.53 npm warn Unknown builtin config "globalignorefile". This will stop working in the next major version of npm. #9 17.53 npm warn Unknown builtin config "python". This will stop working in the next major version of npm. #9 17.79 npm warn Unknown builtin config "globalignorefile". This will stop working in the next major version of npm. #9 17.79 npm warn Unknown builtin config "python". This will stop working in the next major version of npm. #9 18.04 npm warn Unknown builtin config "globalignorefile". This will stop working in the next major version of npm. #9 18.04 npm warn Unknown builtin config "python". This will stop working in the next major version of npm. #9 18.30 npm warn Unknown builtin config "globalignorefile". This will stop working in the next major version of npm. #9 18.30 npm warn Unknown builtin config "python". This will stop working in the next major version of npm. #9 18.30 npm warn Unknown cli config "--unsafe-perm". This will stop working in the next major version of npm. #9 53.71 npm warn deprecated boolean@3.2.0: Package no longer supported. Contact Support at https://www.npmjs.com/support for more info. #9 53.75 npm warn deprecated har-validator@5.1.5: this library is no longer supported #9 54.06 npm warn deprecated lodash.get@4.4.2: This package is deprecated. Use the optional chaining (?.) operator instead. #9 54.82 npm warn deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142 #9 58.33 npm warn deprecated uuid@3.4.0: Please upgrade to version 7 or higher. Older versions may use Math.random() in certain circumstances, which is known to be problematic. See https://v8.dev/blog/math-random for details. #9 60.19 npm warn deprecated node-red-dashboard@3.6.6: No longer maintained - please look at alternatives such as @flowfuse/node [... cut ...] rm --fetch-timeout=300000 --fetch-retry-mintimeout=20000 --fetch-retry-maxtimeout=120000 && npm rebuild --build-from-source @serialport/bindings-cpp && npm cache clear --force && echo -e \"StrictHostKeyChecking no\" >> /etc/ssh/ssh_config && patch -d /opt/node_modules/node-red-dashboard -p1 -i /tmp/node-red-dashboard-show-dashboard.patch && apk del --no-cache --purge .build-dependencies && rm -fr /etc/nginx /root/.cache /root/.npm /root/.nrpmrc /tmp/*" did not complete successfully: exit code: 1

VinceCvrr, 2025-08-10:
I ve got the same error

AURELIODINVERNO, 2025-08-10:
yes, same here...

jean-luc1203, 2025-08-10:
thank you for your feedback. I can see that there is a piece of dependency that is now depreciated.

`npm warn deprecated node-red-dashboard@3.6.6: No longer maintained`

I'll look into it tomorrow if possible. No idea yet how long it will take. I'll let you know when it's OK

AURELIODINVERNO, 2025-08-10:
Merci beaucoup et bonne soirée !

npm warn deprecated ***@***.***: No longer maintained

lgrenetier, 2025-08-12:
I was able to install the addon by changing 
    "node-red-dashboard": "*", to "node-red-dashboard": "3.6.3",
in the dependencies section of the package.json file

jean-luc1203, 2025-08-12:
Great news, I'll test it tonight or tomorrow. Thank you very much for your help.👍
That would suit me fine while I wait to redo the module more neatly. Since I no longer use the Node Red dashboard, I need to remove everything related to it.

jean-luc1203, 2025-08-12:
It's already available to try. I had misread your message

AURELIODINVERNO, 2025-08-12:
Great thanks!

Aurelio D'Inverno

AURELIODINVERNO, 2025-08-12:
ok, it seems to have installed, will try tomorrow to run it.
Thanks!

AURELIODINVERNO, 2025-08-13:
hello again, the add-on has installed successfully and i was able to start
it. however i have an error in the log "

[serialconfig:6e1abcbcc6ebb714] serial port /dev/ttyUSB0 error: Error:
Error: No such file or directory, cannot open /dev/ttyUSB0

can you please help ? (I'm a bit new to linux / ubuntu...) i tried to use
the SUBSYSTEM command with product id and vendor id but the command is not
recognized ...

Thanks a ton for your help !

Aurelio

AURELIODINVERNO, 2025-08-13:
> Hello Aurelio,
>
> You really need to enter the correct information for the USB port used in
> the correct format. If you use the ttyUSBx format, this may change when you
> reboot or insert another device in the USB port, so you need to go to the
> "hardware" menu and find the correct way of defining the port in the module
> configuration.
> image.png (view on web)
>
> *so copy and paste the ID part*
> image.png (view on web)
>
> in the configuration file to obtain a YAML file that looks like this:
> image.png (view on web)

great, thanks, will try soon!

AURELIODINVERNO, 2025-08-13:
ok... getting further but still i have an issue :

13 Aug 19:41:13 - [error] [serialconfig:6e1abcbcc6ebb714] serial port
/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A5069RR4-if00-port0 error:
Error: Error: No such file or directory, cannot open
/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A5069RR4-if00-port0

maybe an authorization issue?

the file is there for sure :-)

any idea?

thanks a ton - and sorry to bother you ...

Aurelio

AURELIODINVERNO, 2025-08-14:
> thanks a ton - and sorry to bother you ...
> No problem.
>
> Can you put here the screenshots of
>
>    - The hardware part with the USB tty* adapters
>    - The JKBMS module start-up log (hiding the MQTT password)
>    - The YAML of the module configuration (press the 3 small vertical
>    dots at the top right and then display in YAML
>
> This might help. Hope so

here some screenshots and logs

tell me please if something is missing.

Thanks in advance for your  feedback

Aurelio

jean-luc1203, 2025-08-14:
your screenshots don't seem to have gone through, I can't see any attachments or images

AURELIODINVERNO, 2025-08-14:
trying again...

AURELIODINVERNO, 2025-08-14:
one more screenshot

AURELIODINVERNO, 2025-08-15:
one more screenshot. ttyUSB0 does not show up in the list of hardware...

[image]

jean-luc1203, 2025-08-15:
Now things are clear.
You're not using the Home Assistant OS (HAOS) but the Supervised or even Core version? Correct?
My software **is intended for the HAOS** version, because that's what manages the Docker containers with the necessary rights to access the USB ports.
In your version, you clearly need to give the container access rights to the ttyUSBx ports and, above all, you need to declare in the Dockerfile that the physical port (ttyUSB0), for example, is to be brought up into de container. 
That's why it doesn't see it. **It's not automatic like HAOS**!

If possible, I'd advise you to go straight to a physical or virtual machine (VM) with HAOS installed on it. It'll work immediately.

AURELIODINVERNO, 2025-08-15:
Ok, busy now, will get back to it in a bit and let you know. Thanks for
your time.

Aurelio D'Inverno

AURELIODINVERNO, 2025-08-15:
Jean-Luc, thanks for the thoughts, however i DID install HAOS through the official channel (before i had a version within Ubuntu - but it did not have all the capabilities). here the screenshot with details. Can you suggest the details of the commands to use (i'm not a Linux expert :-)). Thanks!

[image]

AURELIODINVERNO, 2025-08-15:
i have installed HAOS on a **dedicated physical machine** - with disk restore - as suggested by the official HA website

AURELIODINVERNO, 2025-08-15:
the USB to JKBMS connection is on this other Ubuntu machine

AURELIODINVERNO, 2025-08-15:
more info : installation type

[image]

jean-luc1203, 2025-08-15:
> the USB to JKBMS connection is on this other Ubuntu machine

The adapter is not on the HAOS machine running my module ?
I don't understand the structure of your installation.
From what I can see in the screenshot, HAOS does not detect the USB port.
It must be physically connected to this machine. Let me know. We'll get there 💪
Retirees do a great job 😚

AURELIODINVERNO, 2025-08-15:
mhh... maybe A drawing woudl help ... (thanks to samsung)... your module is loaded on PC2

[image]

jean-luc1203, 2025-08-15:
Okay, loaded into PC2?
What's in PC2 then?
For me, everything should be in PC1.
And if that's not possible, we'll have to use a TCP/IP gateway -- rs485.

AURELIODINVERNO, 2025-08-15:
Well, on PC1 we have HAOS which is accessed through
homeassistant.local:8123.

Should the usb be connected to PC1 ? Where HAOS is running?

Aurelio D'Inverno

AURELIODINVERNO, 2025-08-15:
ok, tried and now i have a different behavior. something seems to be moving...

[image]

jean-luc1203, 2025-08-15:
Yes, it looks promising. 
Now, if you haven't already done so, you need to install the **Mosquitto broker add-on** on this pc1 machine. Install and configure th**e MQTT integration** that will connect to this broker.
 My module goes through this broker to send ans t'écrive all the data to HAOS. This allows it to be used from anywhere and with any software that supports MQTT.

jean-luc1203, 2025-08-15:
Simply use the username and password of the haos user you are logged in as.
The other two software programs do not require any special configuration.

AURELIODINVERNO, 2025-08-16:
ok, tried to install the dashboard and i'm getting this, i suspect some config error.

[image]

jean-luc1203, 2025-08-16:
Aurelio, I've created a new topic just for the dashboard setup. I can close this one, which I think is finished, if you don't mind.
It will be easier to read for people who want to use this dashboard.