jean-luc1203/jkbms-rs485-addon #21: Debugging

brycelarge, 2025-08-14:
I would like some assistance on how to turn on debugging or a more detailed debugging mode? 

I setup an ESP32 to create a wifi RS485 bridge and Im trying to debug the connection and Im not sure how to.

On the ESP32 device I can see the incoming connection, it connects then immediately disconnects:
07:03:14	[D]	[sensor:104 'Modbus Client Count': Sending state 1.00000  with 0 decimals of accuracy
07:03:15	[D]	[stream_server:163] Client 192.168.200.6 disconnected

But on this software I cant seem to access any debug information on the gateway connection: if I change to a non existiant IP address of the gateway I get a failed tcp request in the log: 14 Aug 07:03:39 - [error] [tcp request:374bbe6e8c64a12a] connect failed 192.168.200.133:502. With a working IP i get nothing in the logs but also nothing in MQTT about any of the BMS in the RS485 network.

In your documentation you detail log_level but adding log_level: trace or log_level: debug shows nothing.

Here is the ESP32 wifi bridge yaml template I am using if you are interested, I might be going down the wrong road here:

substitutions:
  name: jkbms-modbus-gateway
  friendly_name: "JKBMS Modbus Gateway"
  tx_pin_uart: GPIO17
  rx_pin_uart: GPIO16
  talk_pin_rs485: GPIO4
  baud_rate: 115200

# Basic ESPHome configuration
esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  comment: "JKBMS RS485 Modbus WiFi Gateway"

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  level: DEBUG
  baud_rate: 0

external_components:
  - source: github://oxan/esphome-stream-server@main
    components: [stream_server]

# Enable Home Assistant API
api:
  encryption:
    key: "YOUR_API_ENCRYPTION_KEY_HERE"

# Enable OTA updates
ota:
  - platform: esphome
    password: "YOUR_OTA_PASSWORD_HERE"

# WiFi connection
wifi:
  ssid: "YOUR_WIFI_SSID_HERE"
  password: "YOUR_WIFI_PASSWORD_HERE"
  # Enable fallback hotspot in case WiFi connection fails
  ap:
    ssid: "${name}_fallback"
    password: "YOUR_FALLBACK_PASSWORD_HERE"

output:
  - platform: gpio
    pin: ${talk_pin_rs485}
    id: rs485_dir_pin

# UART configuration for RS485
uart:
  - id: uart_bus
    baud_rate: ${baud_rate}
    tx_pin: ${tx_pin_uart}
    rx_pin: ${rx_pin_uart}
    rx_buffer_size: 2048

# Stream server configuration
stream_server:
  uart_id: uart_bus
  port: 502
  buffer_size: 2048

# Add connection status sensor
binary_sensor:
  - platform: stream_server
    connected:
      name: "Modbus Client Connected"

# Add connection count sensor
sensor:
  - platform: stream_server
    connection_count:
      name: "Modbus Client Count"
  - platform: uptime
    name: Uptime Sensor

# Enable web server for easy access
web_server:
  port: 80

# Add a restart button
button:
  - platform: restart
    name: "Restart Bridge"

brycelarge, 2025-08-14:
I just want to see whats happening on your code side when connecting to the ESP32 so I can debug the connection, currently your logs show nothing so I know its not connected but why whats wrong?

I'm using the following:
Generic ESP32 or ESP8266 board
SERIAL TTL to RS485 converter

This is the setup I have
[image]

jean-luc1203, 2025-08-14:
I'll take a look at this tonight to give you more information about the dialogue.
In the meantime, you can connect via the Android WiFi Serial Controller app. Connect to your IP and port and watch the communications live. Take a look here, I talk about it: https://domosimple.eu/forum/thread-784-post-7678.html#pid7678

What about the uart direction ?

```
uart:
  id: uart_bus
  baud_rate: ${baud_rate}
  tx_pin: ${tx_pin_uart}
  rx_pin: ${rx_pin_uart}
  rx_buffer_size: 2048
  # Ajoutez cette ligne pour gérer la direction RS485
  direction_pin: 
    number: ${talk_pin_rs485}
    inverted: false

```

jean-luc1203, 2025-08-15:
Ah, that confirms that there is indeed a discussion. 
I notice that the response seems to be cut off. The response buffer is 308 bytes long.
It starts with  55 aa eb 90 .
It would also be a good idea to check the timing so that the entire response is in a single frame

jean-luc1203, 2025-08-15:
Yes !!!! 👍 
Once it's been tested, can you post some photos of your equipment?
I think we could integrate it into this GitHub, what do you think? 
It's a nice, inexpensive solution that's integrated with Esphome and HAOS.
You could also start a thread on my forum with photos and videos to explain your work. I'd really appreciate it.

brycelarge, 2025-08-15:
Yes it’s working but not very well I think there is still a long way to go. 

You can share the solution on your project for others to use.

The buffer size is too big, just the value I had at the time I noticed it start to work.

closed 2025-08-22 by brycelarge

jean-luc1203, 2025-08-22:
@brycelarge 
I ordered some RS485 adapters.
I'm going to try assembling them with the ESP32 soon to see how it goes.
If you've made any improvements to the code, I'd love to hear about them.