jean-luc1203/jkbms-rs485-addon #22: Customized dashboard from Ko-fi.com/smartphoton does not work immediately

jean-luc1203, 2025-08-16:
> ok, tried to install the dashboard and i'm getting this, i suspect some config error.
> 
> [image]
>  

 _Originally posted by @AURELIODINVERNO in [#20](https://github.com/jean-luc1203/jkbms-rs485-addon/issues/20#issuecomment-3193677091)_

AURELIODINVERNO, 2025-08-16:
What's at home you mean the batteries  and their cabling to the HA server?

**here the yaml code**
views:
  - type: sections
    max_columns: 2
    title: JK-BMS Live
    path: jk-bms-live
    sections:
      - type: grid
        cards:
          - type: custom:mushroom-template-card
            primary: Charge
            secondary: |-
              {% if states('sensor.bms_1_switch_charge') == '0' %}
                Off
              {% elif states('sensor.bms_1_switch_charge') == '1' %}
                On
              {% else %}
                Inconnu
              {% endif %}
            icon: |-
              {% if states('sensor.bms_1_switch_charge') == '0' %}
                mdi:toggle-switch-off
              {% elif states('sensor.bms_1_switch_charge') == '1' %}
                mdi:toggle-switch
              {% else %}
                mdi:help-circle
              {% endif %}
            icon_color: |-
              {% if states('sensor.bms_1_switch_charge') == '0' %}
                red
              {% elif states('sensor.bms_1_switch_charge') == '1' %}
                green
              {% else %}
                grey
              {% endif %}
            entity: sensor.bms_1_switch_balance
            grid_options:
              columns: 4
              rows: 1
            tap_action:
              action: more-info
          - type: custom:mushroom-template-card
            primary: Décharge
            secondary: |-

**here the mqtt screen** :
[image]

jean-luc1203, 2025-08-17:
Let's sleep on it. I think I've figured out what's wrong.

In this dashboard, I'm using display cards **that aren't included** by default in HAOS.
They can **be installed via HACS** (the HAOS self-service store). To do this, follow this tutorial (3 minutes).
Here's an example in [french](https://www.domo-blog.fr/comment-installer-hacs-home-assistant-store-integrations-custom-ha/)
Once installed, you need to install the card:

- **Mushroom**

[image]

- **bar-card** (I've read that this will no longer be supported early next year)

[image]

- **button-card**

[image]

- **stack-in-card**

[image]

All these cards are to be added to HAOS. They are not related to my module.

AURELIODINVERNO, 2025-08-17:
Ok will try a bit later

AURELIODINVERNO, 2025-08-17:
well... i downloaded the cards (except the bar-card, it's not there...) and now i'm getting a better picture but still not good...

[image]

AURELIODINVERNO, 2025-08-17:
strangely i have 3 BMS 

and in BMS_1 the value is not even filled... mapping problem perhaps?

[image]
[image]

AURELIODINVERNO, 2025-08-20:
Jean Luc,

j'ai dû réinstaller le Mqtt broker, il y avait un problème d'accès. Maintenant ton module s'arrête à l'ouverture du port - est-ce normal? 
Screenshot 2025-08-20 114847.png

Par ailleurs le mqtt broker semble se comporter d'une façon bizarre (closed ports) 
Screenshot 2025-08-20 115423.png

 et je ne vois aucune device dans la liste par contre je vois la liste des devices dans le MQTT Explorer 
Screenshot 2025-08-20 115545.png

une idée?
Merci

[image]
[image]
[image]

AURELIODINVERNO, 2025-08-20:
bon, apres un peu de bidouillage j'ai les devices qui sont apparues (BMS Master uniquement). 

[image]

AURELIODINVERNO, 2025-08-20:
these are read only values, right? i have a Master/Slave config, cannot change that ... thanks for confirming.

jean-luc1203, 2025-08-20:
Yes, if one of the bms is in position 0000, it becomes the bus master and broadcasts its data.  It aggregates the data, in particular the sockets, to inform the inverter.  In this case, it's impossible to modify parameters remotely via the cable. You have to use Bluetooth to do this. 
I can't remember your configuration. Is it your inverter that asks for a bms to be in position 0000? If not, you need to set addresses 1 and 2 on your bms and then you'll have much more data, especially all the data from each bms separately. 
So it works again?

jean-luc1203, 2025-08-21:
Aurelio, did you define master-slave because your inverter communicates via rs485 with the JKBMS ?

AURELIODINVERNO, 2025-08-21:
https://www.youtube.com/watch?v=BqZO67-npyo

master defines parameters across the array.

jean-luc1203, 2025-08-21:
Personally, I'm not really sure whether there's a conversation between each BMS. 
Do you have more information on this point? What the master BMS does, in my opinion, is to aggregate the SOCs of the other BMSs and supply them to the inverter. 
It makes sure to show only one pack. But from there to the Master BMS controlling the other BMS I have a big doubt.
I'd love to hear from someone who reads us and knows what's going on.
I'm **obviously going to watch the video** you linked to me later today, and I'll be back to discuss it here.

AURELIODINVERNO, 2025-08-21:
[image]

AURELIODINVERNO, 2025-08-25:
Hello,
i had a couple of days without internet and today it's back. 

Gave me a chance to check what's coming out of your module, and i had a different picture than before (without me changing anything).... not sure why. here the pic of what i'm getting, now i have 3 BMSs... any idea what's going on?

Thanks!

[image]

jean-luc1203, 2025-08-25:
Hello Aurelio.
Could you stop the module? 
Go to the MQTT devices and delete all BMS.
Then, if you can, use MQTTexplorer to connect to the MQTT broker and also delete all BMS topics. Just take the primary BMS key and it will delete all subtopics. 
So now there should be nothing left.
Restart the module and show me the module logs and the devices it discovers.
What mode are you in? Broadcasting or master-slave?
We'll take stock of the situation afterwards.

AURELIODINVERNO, 2025-08-25:
just done, here is what i get (config = bms_broadcasting = false

[image]
[image]

jean-luc1203, 2025-08-25:
I don't understand. It looks like you have installed the module several times.
Look at the titles, one of them is different. 
IP gateway and not the other two. 
We agree that there are only two BMS.
Did you do the latest updates correctly?
Can you show me the installed modules? Can you uninstall all the modules and install the latest version, and then show me what's going on? Did the devices recreate themselves? You didn't create them manually, did you?
Can you also show me the entire configuration file in yaml format? Hide the password.

closed 2025-08-27 by jean-luc1203